adafruit/Adafruit_Python_PN532 #1: Mifare 7 byte UID

bimmel1231, 2015-06-10:
Any hint how to read a 7 byte UID of a mifare ultralight tag?

m-hofmann, 2016-01-03:
Hello,

sorry for responding to such an old thread, but I have just encountered the same problem. Hopefully my solution will help someone.

There seems to be a bug in the `read_passive_target()` method that restricts it to reading 4 byte cards only. If you turn on logging, you will see that the 7 byte UID response frame returned from the PN532 to your Raspberry Pi ends in a Mifare ATS sequence (0x...067577810280). The packet checksum and the postamble will not be read. Therefore, checksum calculation will fail.

One simple solution is to change
```
response = self.call_function(PN532_COMMAND_INLISTPASSIVETARGET,
                                      params=[0x01, card_baud],
                                      response_length=17)
```
to 
```
response = self.call_function(PN532_COMMAND_INLISTPASSIVETARGET,
                                      params=[0x01, card_baud],
                                      response_length=19)
```
This increases the frame length that is read from the PN532. Thus, the checksum and the postamble will be read. One might argue that the implementation here is a bit sloppy and that the number of read bytes should depend on the packet length returned by the PN532. However, doing so would mean to re-write large parts of the driver, so I recommend the quick and dirty solution for the time being.

You may wonder if reading more bytes may lead to errors for other card types (with a 4 byte UID). In fact, it seems that the PN532 pads its message frames with trailing 'a' characters to prevent such problems. These trailing characters are ignored by the frame handling code, so you are safe there.

Hope it helps!
Regards

Martin

ladyada, 2018-08-25:
hiya - we've re-written this library to support circuitpython/python, i2c and uart as well! your fix was hand added there. thanks! https://github.com/adafruit/Adafruit_CircuitPython_PN532

closed 2018-08-25 by ladyada