adafruit/Adafruit_Python_PN532 #2: Add support for NTAG203, etc. tags

tdicola, 2015-06-30:
The Arduino library supports these tags so we should add them to the Python library too.  This is a TODO to take a look at adding the support in the future.

coenroest, 2016-03-10:
Any progress on this issue?

igorvukotic, 2016-03-10:
nope..

coenroest, 2016-06-17:
Any timeline for this issue?
I would love to work with other NFC tags than Mifares.

Giovasci, 2017-04-25:
Hello! There is something new about the possibility to Read/Write tags NTAG2xx using a Raspberry Pi?

monsdar, 2018-08-25:
Another year passed... Problably no updates on this I guess? :(

ladyada, 2018-08-25:
hiya we're deprecating this library and moving to a circuitpython one which supports i2c/spi/uart. its 99% same as this one but will work across more platforms!
https://github.com/adafruit/Adafruit_Circuitpython_PN532
if you can send a pull request for ntag support there, we'd love to add it!

closed 2018-08-25 by ladyada