adafruit/Adafruit_Python_PN532 #3: Support for I2C and/or UART

araid, 2015-12-10:
What would it take to add support for the other 2 protocols?
Is there an application note where the commands for I2C or UART are specified?

It would be very nice to be able to use this with 4 wires instead of 6 :)

ladyada, 2015-12-10:
hiya uart is supported in libnfc
https://learn.adafruit.com/adafruit-nfc-rfid-on-raspberry-pi/testing-it-out

i2c is not supported because pn532 uses clockstretching, so it doesnt work on raspberry pi and probably a bunch of other devices

closed 2015-12-10 by ladyada

araid, 2015-12-11:
Yes, libnfc works great, but I want to control the pn532 from Python so I just wondered if you had any tips for adding another protocol to your library. It would be simpler to use than having to install libnfc, finding bindings for it, etc.

I didn't know about clock stretching. Someone apparently made a library using i2c but I haven't been able to get it to work:
https://github.com/HubCityLabs/py532lib
https://web.archive.org/web/20140919081033/http://hubcitylabs.org/nfc-on-raspberrypi-with-pn532-py532lib-and-i2c/

Thanks anyway!

ladyada, 2015-12-11:
im...a little dubious that the py532 library would work, im not surprised if it doesnt! :(