adafruit/Adafruit_Python_PN532 #6: PN532 class using hardware SPI

sfranzyshen, 2016-02-05:
How can you open the hardware spi (spidev) what is the format for the spi parameter is passed. 

From the Source code:

def __init__(self, cs, sclk=None, mosi=None, miso=None, gpio=None, 
                 spi=None):
        """Create an instance of the PN532 class using either software SPI (if 
        the sclk, mosi, and miso pins are specified) 

> or hardware SPI if a
>         spi parameter is passed.  

The cs pin must be a digital GPIO pin.
        Optionally specify a GPIO controller to override the default that uses 
        the board's GPIO pins.
        ""

sfranzyshen, 2016-02-06:
In the code ... it seams incomplete ... if you set the spi=spi (or anything other than None) it then never sets up Adafruit_GPIO.SPI?

        # Setup SPI provider.
        if spi is not None:
            logger.debug('Using hardware SPI.')
            # Handle using hardware SPI.
            self._spi = spi
            self._spi.set_clock_hz(1000000)
        else:
            logger.debug('Using software SPI')
            # Handle using software SPI.  Note that the CS/SS pin is not used
            # as it will be manually controlled by this library for better
            # timing.
            self._spi = SPI.BitBang(self._gpio, sclk, mosi, miso)

sfranzyshen, 2016-02-06:
I think I understand ... Do I have to setup the SPI (Adafruit_GPIO.SPI) in my code then call ...
pn532 = PN532.PN532(spi=PN532.SPI)
pn532.begin()

tdicola, 2016-02-08:
Yep you have the right idea about using hardware SPI, here's an example from a different chip but you can see the difference between using software bit banged SPI and hardware SPI is passing in a spi paramter that's constructed from the port & device: https://github.com/adafruit/Adafruit_Python_MCP3008/blob/master/examples/simpletest.py#L12-L22

For hardware SPI the CS pin will depend on the hardware.  For the Raspberry Pi for example SPI port 0 and device 0 is going to use GPIO pin CE0 / GPIO8, and SPI port 0 device1 will use CE1 / GPIO7.  These devices correspond to the /dev/spidev*.* devices, like /dev/spidev0.0 for port 0, device 0 and /dev/spidev0.1 for port 0, device 1.  The CS pin is needed with SPI because it tells the chip that it's about to receive a command from the SPI bus (so in theory multiple devices can share the bus as long as they have unique CS pins).  Sending and receiving data with the library through those /dev/spidev devices will automatically take care of toggling the CS pin when appropriate.

If you're running into trouble with the code posted above can you clarify what hardware you're using and how it's wired to the device?  Check that the hardware CS line for port 0 and device 0 is what you have connected to the PN532's CS line.  Also make sure you've adjusted the jumpers on the PN532 to enable SPI, see the picture of them here: https://learn.adafruit.com/raspberry-pi-nfc-minecraft-blocks/hardware-wiring

closed 2016-02-08 by tdicola

sfranzyshen, 2016-02-10:
Here is the code I have ...

```
import Adafruit_GPIO.SPI as SPI
import Adafruit_PN532 as PN532

# Configuration for Raspberry Pi hardware SPI:
#CS         = 8
CS         = 24
SPI_PORT   = 0
SPI_DEVICE = 0

# Create an instance of the PN532 class.
pn532 = PN532.PN532(cs=CS, spi=SPI.SpiDev(SPI_PORT, SPI_DEVICE))

# Call begin to initialize communication with the PN532.  Must be done before
# any other calls to the PN532!
pn532.begin()
```
and here is what gets returned ...
```
Traceback (most recent call last):
  File "readmifare.py", line 55, in <module>
    pn532.begin()
  File "build/bdist.linux-armv6l/egg/Adafruit_PN532/PN532.py", line 336, in begin
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/GPIO.py", line 64, in set_low
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/GPIO.py", line 194, in output
IOError: [Errno 22] Invalid argument
```
Please don't close issues ... when they are not closed!

Carl0701, 2017-05-21:
Hello sfranzyshen:

I am in the same situation as you:

`IOError: [Errno 22] Invalid argument`

Any solution...?

dzyk, 2017-09-20:
i have this problem too