adafruit/Adafruit_Python_PN532 #7: PN532 class using hardware SPI

sfranzyshen, 2016-02-10:
Here is the code I have ...
```
import Adafruit_GPIO.SPI as SPI
import Adafruit_PN532 as PN532

# Configuration for Raspberry Pi hardware SPI:
#CS         = 8
CS         = 24
SPI_PORT   = 0
SPI_DEVICE = 0

# Create an instance of the PN532 class.
pn532 = PN532.PN532(cs=CS, spi=SPI.SpiDev(SPI_PORT, SPI_DEVICE))

# Call begin to initialize communication with the PN532.  Must be done before
# any other calls to the PN532!
pn532.begin()
```
and here is what gets returned ...
```
Traceback (most recent call last):
  File "readmifare.py", line 55, in <module>
    pn532.begin()
  File "build/bdist.linux-armv6l/egg/Adafruit_PN532/PN532.py", line 336, in begin
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/GPIO.py", line 64, in set_low
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/GPIO.py", line 194, in output
IOError: [Errno 22] Invalid argument
```
Please don't close issues ... when they are not closed

coenroest, 2016-03-11:
I run in to the same error.
How to setup this library with hardware SPI?

Carl0701, 2017-05-24:
I am in the same situation.
Any ideas..?

bluesign2k, 2017-07-24:
Apologies in advance for the large wall of text, but the information about why this problem occurs and how to get around it seems to be a little scattered over the internet. Hopefully bringing my findings together will help some others progress with their projects.

Anyway, I came across this problem this weekend and have now managed to get it working (sort of). There's actually a number of ways to do it, some more elegant than others. The method I tried first and now have working is probably the less elegant, but it works...

Some background to the problem...
The RPi SoC has three hardware SPI modules in total, SPI0, SPI1, and SPI2. The latter two are two of the three auxiliary devices on the SoC. All this information can be found in the [BCM2835 ARM Peripherals datasheet](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf). While some models of the RPi use the BCM2836 and BCM2837, this document is (apparently) still correct for those devices.

The 'default' device, and the one you'll see listed if you do an `ls /dev/spi*` on a Pi with a standard setup is SPI0.... and this is where the problem lies. While there a quite a number of ways to setup this module, of the few things it can't do, LSBFIRST is one (check page 153 of the datasheet to see the available options of SPI0's control register). Therefore, assuming you're working with a Raspberry Pi with a standard setup, you can use the hardware SPI module, but you will have to handle the conversion to lsbfirst in software.

Manually bit-reversing every byte that's both sent and received isn't hard, but it can be costly time-wise. Personally, this whole exercise began by me wanting to learn Python and my impression so far is that it's great at handling data but awful at speed. Therefore, my solution was to hack the Python SPI module that interfaces to the kernel ([py-spidev](https://github.com/doceme/py-spidev)) because it's written in C. In there I ended up modifying SpiDev_writebytes, SpiDev_readbytes and SpiDev_xfer2 to use a LUT to reverse the bits if SPI_LSB_FIRST was set (honestly, it's nothing more than `if(self->mode & SPI_LSB_FIRST) {buf[ii] = lsb_first_table[buf[ii] & 0xFF];}` where ` lsb_first_table` is just a 256 byte table of bit-reversed bytes) since each of those functions already iterates through the entire rx/tx buffers anyway. I also commented out the call to set the mode in SpiDev_set_lsbfirst. This all works fine for our purpose here on an RPi, but clearly it's hacky and isn't a clean enough solution for a PR on spidev as it will probably break things for other platforms.

The second (and probably better) solution is to enable and use one of the two aux SPI modules on the Pi as both of these *do* support hardware LSBFIRST (see the control register tables on pages 22/23/24 of the datasheet). I haven't tried this yet, but I believe adding `dtoverlay=spi1-3cs` to `/boot/config.txt` should enable the second module. However, I believe for some RPi models there is then another problem. With the Pi3 (and therefore probably the Pi Zero W too), the bluetooth module uses UART0 and the default serial console is then pushed to UART1. UART1 is the third auxiliary device and shares some resources with SPI1 (and SPI2). Currently, enabling both at the same time causes a conflict and will crash your Pi. The two solutions are therefore to either disable bluetooth (by adding `dtoverlay=pi3-disable-bt` to `/boot/config.txt`) so that the console is configured on UART0 or to leave bluetooth enabled on UART0 and disable the console by removing `console=serial0,115200` from `/boot/cmdline.txt`.

Anyway, I hope that helps a little...

Carl0701, 2017-07-24:
Thank you for the contribution. I will continue testing ...

Zontex, 2018-01-31:
I tried the dtoverlay=pi3-disable-bt to /boot/config.txt solution but no luck, anyone got this issue fixed?

ladyada, 2018-08-25:
hiya - we've re-written this library to support circuitpython/python, i2c and uart as well! it now also supports hardware SPI using Adafruit_blinka circuitpython API

check it out here https://github.com/adafruit/Adafruit_CircuitPython_PN532

closed 2018-08-25 by ladyada