adafruit/Adafruit_Python_PN532 #11: TypeError: argument 1 must be string, not int

wimg, 2017-06-12:
- Platform/operating system : Raspberry Pi with Raspbian operating system
- Python version (run `python -version` or `python3 -version`):  2.7.9
- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "mcpi_listen.py", line 70, in <module>
    pn532 = PN532.PN532(cs=CS, sclk=SCLK, mosi=MOSI, miso=MISO)
  File "build/bdist.linux-armv7l/egg/Adafruit_PN532/PN532.py", line 169, in __init__
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/GPIO.py", line 278, in setup
    pull_up_down=self._pud_mapping[pull_up_down])
TypeError: argument 1 must be string, not int
- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Running example readmifare.py or mcpi_listen.py (sudo python filename.py)

Any idea what might be wrong ?

BRJMM, 2018-02-16:
I have the same problem, but with the "readmifare.py" file.

BRJMM, 2018-02-16:
I have the same error.

ladyada, 2018-08-25:
hiya - we've re-written this library to support circuitpython/python, i2c and uart as well! we've redone the GPIO code so this should be fixed there 
 thanks! https://github.com/adafruit/Adafruit_CircuitPython_PN532

closed 2018-08-25 by ladyada